chenmozhijin/OpenWrt-K #25: 首先感谢大佬的工作流，真是太好用了，不知道能否修改sh文件增加自己想编译的包

759071793, 2023-07-18:
非常感谢大佬的工作。确实非常好用。自己fork修改和删除一些包非常方便编译。
有一些问题想咨询下是否可以实现。
如果使用sh文件进行配置文件的生成，有些自己想安装的包没有下载，默认下载的是大佬自己的包仓库，不知道能否修改sh文件增加自己想编译的包，否则在sh勾选包的时候由于没有lu-app-包，没有将包编译进去，目前自己只能fork仓库手动修改工作流文件从别的仓库按照大佬的语法从别的仓库复制，然后再手动添加*.config，但是自己对某些包需要用那些附加包不太清除，有可能漏掉一些.config。

759071793, 2023-07-18:
git clone https://github.com/sirpdboy/luci-app-lucky
          git clone https://github.com/sbwml/luci-app-mosdns
          
          cp -r $DOWNLOAD_ROOT_PATH/other/luci-app-mosdns/luci-app-mosdns $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/other/luci-app-mosdns/mosdns $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/other/luci-app-lucky/luci-app-lucky $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/other/luci-app-lucky/lucky $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/lede/luci/applications/luci-app-wireguard $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/lede/luci/applications/luci-app-vsftpd $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/lede/luci/applications/luci-app-syncdial $CMZJ_PACKAGE_ROOT_PATH/
          cp -r $DOWNLOAD_ROOT_PATH/lede/luci/applications/luci-app-ramfree $CMZJ_PACKAGE_ROOT_PATH/
          
          不知道我这样子写有没有问题，我实在是不懂这个openwrt的流程结构

chenmozhijin, 2023-07-18:
应该没有问题，你可以在build1的加载自定义配置并生成配置文件步骤中检查你增加的包是否编译。

chenmozhijin, 2023-07-18:
luci-app-vsftpd的依赖[vsftpd-alt](https://github.com/759071793/OpenWrt-K/blob/575ce25dd4bdafdf76aaa62fdfcbb67b113c00ff/config/network.config#L8C1-L8C28)与openwrt官方的[vsftpd-tls](https://github.com/759071793/OpenWrt-K/blob/575ce25dd4bdafdf76aaa62fdfcbb67b113c00ff/config/network.config#L10C1-L10C28)冲突了

759071793, 2023-07-19:
> luci-app-vsftpd的依赖[vsftpd-alt](https://github.com/759071793/OpenWrt-K/blob/575ce25dd4bdafdf76aaa62fdfcbb67b113c00ff/config/network.config#L8C1-L8C28)与openwrt官方的[vsftpd-tls](https://github.com/759071793/OpenWrt-K/blob/575ce25dd4bdafdf76aaa62fdfcbb67b113c00ff/config/network.config#L10C1-L10C28)冲突了

修改不来，删除掉了。
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a build dependency on 'csstidy/host', which does not exist
WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a build dependency on 'luasrcdiet/host', which does not exist
大佬，我使用官方23.05.0-rc2报错，全新拉的官方包，这些依赖请问你晓得在哪里找不。。

759071793, 2023-07-19:
> 修改不来，删除掉了。 WARNING: Makefile 'package/turboacc/luci-app-turboacc/Makefile' has a dependency on 'luci-lua-runtime', which does not exist WARNING: Makefile [...]

使用这两个命令没有报错了，不知道有没得用，正在编译
./scripts/feeds update -i -f
./scripts/feeds install -a -f

chenmozhijin, 2023-07-19:
这种警告很正常，依赖是有的。因为你的feeds还没安装，只要在build1更新 feeds步骤安装过一遍就不会报了，只要安装build1 feeds步骤没有就行

chenmozhijin, 2023-07-19:
还有luci-app-mosdns已经有[软链接](https://github.com/sbwml/luci-app-mosdns/blob/v5/luci-app-mosdns/po/zh_Hans)了不需要修复，修复反而报错。
```
find: 'package/cmzj_package/luci-app-mosdns/po/zh_Hans/zh-cn': Too many levels of symbolic links
find: 'package/cmzj_package/luci-app-mosdns/po/zh-cn/zh-cn': Too many levels of symbolic links
```
而luci-app-lucky中他的zh_Hans只是一个文本文件，需要把他的文本文件删了在创建软链接
```
rm -rf ./luci-app-lucky/po/zh_Hans
ln -s zh-cn ./luci-app-lucky/po/zh_Hans
```

759071793, 2023-07-19:
> 

收到，正在编译中。再次感谢回复

chenmozhijin, 2023-07-20:
已更新此功能，见 74599ba0d215c0643dcca5d3b572e3f9d7c58e42 60dbb4663f44b54000ef20ae881faf1f70fa0974

closed 2023-07-20 by chenmozhijin