chenmozhijin/OpenWrt-K #41: 编译后Turbo ACC没有设置项

smdx, 2023-10-11:
大佬你好，编译后的固件Turbo ACC里面没有设置项。

rootfs包在PVE LXC下运行
OpenWrt 23.05.0-rc4 r23482-7fe85ce1f2
LuCI openwrt-23.05
branch git-23.236.53405-fc638c8

[image]

chenmozhijin, 2023-10-11:
Turbo ACC通过检测”/lib/modules/内核版本/“中的内核模块决定是否显示对应的设置项。
我看了一下[你编译的固件](https://github.com/smdx/OpenWrt-K/releases/tag/v23.10.11-08(x86-64)-(v23.05.0-rc4))，其中模块都是在的。
我怀疑是有可能是获取内核版本的问题请尝试执行```echo -n $(uname -r)```看看它返回的内核版本是否与”/lib/modules/”的一致

smdx, 2023-10-12:
> Turbo ACC通过检测”/lib/modules/内核版本/“中的内核模块决定是否显示对应的设置项。 我看了一下[你编译的固件](https://github.com/smdx/OpenWrt-K/releases/tag/v23.10.11-08(x86-64)-(v23.05.0-rc4))，其中模块都是在的。 我怀疑是有可能是获取内核版本的问题请尝试执行`echo -n $(uname -r)`看看它返回的内核版本是否与”/lib/modules/”中的一致。 我不了解LXC，但我认为有可能是它导致的。

感谢回复，输入命令行查看内核确实是不一致，LXC用的是PVE的内核。请问该如何解决？谢谢
[image]
[image]

chenmozhijin, 2023-10-12:
修改opewnrt中的"/usr/lib/lua/luci/model/cbi/turboacc.lua"的第一行
```
local kernel_version = luci.sys.exec("echo -n $(uname -r)")
```
改为
```
local kernel_version = luci.sys.exec("echo -n '5.15.132'")
```
或改用虚拟机运行openwrt

smdx, 2023-10-13:
> 修改opewnrt中的"/usr/lib/lua/luci/model/cbi/turboacc.lua"的第一行
> 
> ```
> local kernel_version = luci.sys.exec("echo -n $(uname -r)")
> ```
> 
> 改为
> 
> ```
> local kernel_version = luci.sys.exec("echo -n '5.15.132'")
> ```
> 
> 或改用虚拟机运行openwrt

感谢，在修改后设置项有了，但是开启使用全锥形 NAT 还是显示未运行。

chenmozhijin, 2023-10-13:
比较麻烦，lede使用firewall3目前openwrt官方源码22.03以后使用firewall4没办法直接移植，FullCone-NAT的实现方法也不同。
turboacc插件只是一个开关真正实现其功能的是[它的依赖](https://github.com/chenmozhijin/turboacc/tree/package)
不过你可以使用命令进行功能开关
开
```
uci set firewall.@defaults[0].flow_offloading="1"
uci set firewall.@defaults[0].flow_offloading_hw="1"
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="bbr"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
```
关
```
uci set firewall.@defaults[0].flow_offloading="0"
uci set firewall.@defaults[0].flow_offloading_hw="0"
uci set firewall.@defaults[0].fullcone="0"
uci commit firewall
sysctl -w net.ipv4.tcp_congestion_control="cubic"
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart
```

smdx, 2023-10-13:
> 比较麻烦，lede使用firewall3目前openwrt官方源码22.03以后使用firewall4没办法直接移植，FullCone-NAT的实现方法也不同。 turboacc插件只是一个开关真正实现其功能的是[它的依赖](https://github.com/chenmozhijin/turboacc/tree/package) 不过你可以使用命令进行功能开关 开
>
> ```
> uci set [...]

好的，能用fullcone就行。非常感谢

smdx, 2023-10-16:
> 比较麻烦，lede使用firewall3目前openwrt官方源码22.03以后使用firewall4没办法直接移植，FullCone-NAT的实现方法也不同。 turboacc插件只是一个开关真正实现其功能的是[它的依赖](https://github.com/chenmozhijin/turboacc/tree/package) 不过你可以使用命令进行功能开关 开
>
> ```
> uci set [...]

大佬请教一下
使用你提供的命令行开启fullcone
uci set firewall.@defaults[0].fullcone="1"
uci commit firewall
/etc/init.d/dnsmasq restart
/etc/init.d/firewall restart

在/etc/init.d/firewall restart返回的信息确认已经开启fullcone了
但是在Turbo ACC里面还是显示未运行，这个是什么原因？谢谢

smdx, 2023-10-16:
> Turbo ACC使用`cat "/sys/module/nft_fullcone/refcnt"`命令检查fullcone的运行状态，如果为返回0则判断fullcone已启用。
> 
> 我推测这是因为：fullcone的实现依赖于内核模块[kmod-nft-fullcone](https://github.com/fullcone-nat-nftables/nft-fullcone)这个模块是已经编译到openwrt的内核中了，但是因为LXC没有运行单独的内核，运行在LXC中的openwrt使用的是宿主机（pve）的内核而其中并没有kmod-nft-fullcone这个模块所以就无法使用了。
> 
> 不过还是可以用 [NatTypeTester](https://github.com/HMBSbige/NatTypeTester) 试一它到底有没有用，如果没有用或许只能把openwrt安装到虚拟机中才能解决这个问题
> 
> 相关链接 [关于“/sys/module/”](https://www.kernel.org/doc/Documentation/ABI/stable/sysfs-module) [firewall补丁判断fullcone是否开启的方法](https://github.com/chenmozhijin/turboacc/blob/119adf0a3b14ce0d12b61513ad5774cbcb2f38d4/firewall4-598d9fbb5179667aa0c525040eaa41bc7f2dc015/firewall4/patches/999-01-firewall4-add-fullcone-support.patch#L164C2-L174C5) [lxc项目存储库](https://github.com/lxc/lxc)

使用cat命令查看提示cat: can't open '/sys/module/nft_fullcone/refcnt': No such file or directory
和你说的一样PVE的内核而其中并没有kmod-nft-fullcone这个模块。用过NatTypeTester测试显示FullCone没问题
就不细究了。感谢回复。

closed 2023-10-21 by chenmozhijin

m0eak, 2024-05-26:
> 大佬请教一下 使用你提供的命令行开启fullcone uci set firewall.@defaults[0].fullcone="1" uci commit firewall /etc/init.d/dnsmasq restart /etc/init.d/firewall restart
>
> 在/etc/init.d/firewall restart返回的信息确认已经开启fullcone了 [...]

lxc公用宿主机的内核，宿主机内核没有编译fullcone nat那op也肯定开启不了

smdx, 2024-05-26:
pve内核用的是打了fullcone 模块补丁的，不过已经弃用lxc运行op了。