chenmozhijin/OpenWrt-K #42: 默认仓库好像没用upnp

759071793, 2023-10-20:
大哥你首页写了内置软件，但是我在https://github.com/chenmozhijin/OpenWrt-K/blob/main/config/x86_64/OpenWrt-K/extpackages.config没有看到upnp的包呢

chenmozhijin, 2023-10-20:
这不是拓展软件包呢，openwrt官方的feeds就有这个包https://github.com/openwrt/luci/tree/master/applications/luci-app-upnp

759071793, 2023-10-21:
> 这不是拓展软件包呢，openwrt官方的feeds就有这个包https://github.com/openwrt/luci/tree/master/applications/luci-app-upnp

默认拉取的op在菜单中没有这个，我从[immortalwrt]拉取的好像不起作用，不知道大哥的upnp有没有起作用

chenmozhijin, 2023-10-21:
有的啊？
[image]
[image]

我看你仓库[releases](https://github.com/759071793/OpenWrt-K/releases)的[package.zip](https://github.com/759071793/OpenWrt-K/releases/download/v23.10.21-00(x86-64)-(openwrt-23.05)/package.zip)中元素有这个包的

759071793, 2023-10-21:
> 有的啊？ [image] [image]
> 
> 我看你仓库[releases](https://github.com/759071793/OpenWrt-K/releases)的[package.zip](https://github.com/759071793/OpenWrt-K/releases/download/v23.10.21-00(x86-64)-(openwrt-23.05)/package.zip)中也是有"luci-app-upnp_git-23.293.58403-cc4d9c1_all.ipk"这个包的

好奇怪，这个是我从imortal复制到我的一个op package仓库后再make menuconfig才有upnp的选项，但是默认的op 源码拉取后的application文件夹中我也看到确实是有luci-app-upnp这个包，但是make menuconfig菜单没有选项，太奇怪了，我再检查一下，我现在把我仓库中的luci-app-upnp删除再测试下。
我现在拉取的luci-app-upnp已经编译进固件了，但是这个upnp根本不起作用。拉取的op分支openwrt-23.05

chenmozhijin, 2023-10-21:
试着
```
 ./scripts/feeds update -a
 ./scripts/feeds install -a
```
一下？

759071793, 2023-10-21:
> 试着
> 
> ```
>  ./scripts/feeds update -a
>  ./scripts/feeds install -a
> ```
> 
> 一下？

看来是我的问题，重新拉取后菜单有选项了。
不知道大哥有用过upnp没有，我除了用lede源码的upnp起作用，用官方的编译就没有起作用过

chenmozhijin, 2023-10-21:
没用过upnp也不知道是干什么用的，要是确定真的没法用可以去给openwrt提个issues

759071793, 2023-10-21:
谢谢啦，官方openwrt upnp真有问题。

closed 2023-10-21 by 759071793

wanmyj, 2023-10-22:
> > 试着
> > ```
> >  ./scripts/feeds update -a
> >  ./scripts/feeds install -a
> > ```
> > 
> > 
> > 一下？
> 
> 看来是我的问题，重新拉取后菜单有选项了。 不知道大哥有用过upnp没有，我除了用lede源码的upnp起作用，用官方的编译就没有起作用过

Hi, @759071793 请问一下你是怎么做到使用lede源码的upnp的？

使用lede源码的upnp，这个要到哪里能找到？是可以用来安装的ipk文件吗？还是可以写到云编译环境里(即修改这个repo的config)然后直接出-包含`使用lede源码的upnp`-的升级包？

chenmozhijin, 2023-10-22:
> > > 试着
> > > ```
> > >  ./scripts/feeds update -a
> > >  ./scripts/feeds install -a
> > > ```
> > > 
> > > 
> > > 一下？
> > 
> > 
> > 看来是我的问题，重新拉取后菜单有选项了。 不知道大哥有用过upnp没有，我除了用lede源码的upnp起作用，用官方的编译就没有起作用过
> 
> Hi, @759071793 请问一下你是怎么做到使用lede源码的upnp的？
> 
> 使用lede源码的upnp，这个要到哪里能找到？是可以用来安装的ipk文件吗？还是可以写到云编译环境里(即修改这个repo的config)然后直接出-包含`使用lede源码的upnp`-的升级包？

有是有
https://github.com/coolsnowwolf/luci/tree/master/applications/luci-app-upnp
https://github.com/coledeolsnowwolf/packages/tree/master/net/miniupnpd
但lede是fw3现在的官方源是fw4,应该不兼容。

看了一下官方源的包也还在维护
https://github.com/openwrt/luci/tree/master/applications/luci-app-upnp
是怎么个用不了法？我不了解upnp

wanmyj, 2023-10-23:
@chenmozhijin 感谢回复，upnp对我主要是用来做portforward，我用在qbit下载和打游戏。一个简单的测试办法，关闭路由器的port 3074的 forward（不打游戏应该也没有开），然后到Windows 的 Xbox Networking的设置里（win10在gaming里面，win11没找到这个选项，所以我只在win10下做测试），测试一下连接类型，这时候`NAT Type`应该是`moderate`或者`strict`，打开upnp功能再测试，NAT Type如果变成open，代表unpn是工作的。
[image]