chenmozhijin/OpenWrt-K #44: `TMPDIR` directory is missing when running `config_build_tool.sh`, causing curl returns failure

wanmyj, 2023-10-23:
很好用的workflow！👍

在执行到 `config_build_tool.sh`[的第260行](https://github.com/chenmozhijin/OpenWrt-K/blob/main/config_build_tool.sh#L260)的时候，返回了错误，错误代码是23，原因是 以下代码的`$TMPDIR`目录不存在
```
curl -o $TMPDIR/openwrtext.config -s -L --retry 3 --connect-timeout 20  $DOWNLOAD_URL
```

俺还没有仔细研究这个目录为什么执行到此会不存在。`$(mktemp -d)`是可以正常运行的。发个issue也问问到底是我的环境问题还是脚本问题。

wanmyj, 2023-10-23:
FYI, I am following the instruction of [this wiki 使用OpenWrt‐k配置构建工具
](https://github.com/chenmozhijin/OpenWrt-K/wiki/%E4%BD%BF%E7%94%A8OpenWrt%E2%80%90k%E9%85%8D%E7%BD%AE%E6%9E%84%E5%BB%BA%E5%B7%A5%E5%85%B7)

chenmozhijin, 2023-10-23:
能`bash -x ./config_build_tool.sh`运行一下吗？
我这里测试的没有问题

wanmyj, 2023-10-23:
bash 脚本确实不太能看明白，所以我直接贴过来了，麻烦大佬多费费眼睛。。。
<details>
  <summary>Bash log</summary>

```
sh@sh-vm:~/openwrt_config_gen$ bash -x ./config_build_tool.sh
+ trap 'rm -rf "$TMPDIR"' EXIT
++ mktemp -d
+ TMPDIR=/tmp/tmp.h55hLf1ZTg
+ start
+ install_dependencies
+ type which
+ echo 检测运行环境
检测运行环境
+ which apt-get
+ PM=apt
+ which yum
+ n=0
+ case $PM in
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^build-essential'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ grep -c '^clang'
++ apt list --installed
+ '[' 2 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^flex'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^bison'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^g++'
+ '[' 4 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^gawk'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^gcc-multilib'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^g++-multilib'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^gettext'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^git'
+ '[' 2 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^libncurses-dev'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^libssl-dev'
+ '[' 1 -eq 0 ']'
+ for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
++ apt list --installed
++ grep -c '^python3-distutils'
[... 492 lines cut ...]
  }
]'
+ grep -q '"name": "x86_64"'
+ break
+ '[' -e buildconfig.config ']'
+ [[ -e /tmp/tmp.h55hLf1ZTg/openwrtext.config ]]
+ DOWNLOAD_URL=https://raw.githubusercontent.com/chenmozhijin/OpenWrt-K/main/config/x86_64/OpenWrt-K/openwrtext.config
+ echo DOWNLOAD_URL is https://raw.githubusercontent.com/chenmozhijin/OpenWrt-K/main/config/x86_64/OpenWrt-K/openwrtext.config and TMPDIR is /tmp/tmp.h55hLf1ZTg
DOWNLOAD_URL is https://raw.githubusercontent.com/chenmozhijin/OpenWrt-K/main/config/x86_64/OpenWrt-K/openwrtext.config and TMPDIR is /tmp/tmp.h55hLf1ZTg
+ '[' '!' -d /tmp/tmp.h55hLf1ZTg ']'
+ echo tempdir non exist
tempdir non exist
+ curl -o /tmp/tmp.h55hLf1ZTg/openwrtext.config -s -L --retry 3 --connect-timeout 20 https://raw.githubusercontent.com/chenmozhijin/OpenWrt-K/main/config/x86_64/OpenWrt-K/openwrtext.config
+ exitstatus=23
+ '[' 23 -ne 0 ']'
+ echo curl exe exitstatus is 23
curl exe exitstatus is 23
+ whiptail --title 错误 --msgbox 'OpenWrt-K拓展配置下载失败，下载链接：\nhttps://raw.githubusercontent.com/chenmozhijin/OpenWrt-K/main/config/x86_64/OpenWrt-K/openwrtext.config,curl返回值：0' 10 110
+ return 6
+ exitstatus=6
+ '[' 6 -eq 6 ']'
+ exit 0
+ rm -rf /tmp/tmp.h55hLf1ZTg
```
</details>

chenmozhijin, 2023-10-23:
我拿你的跟我的对比了一下，前面基本上都是一样的
<details>
    <summary>Bash log</summary>

    ```
    cmzj@DESKTOP-2CSNPOQ:~/project/test$ bash -x ./config_build_tool.sh
    + trap 'rm -rf "$TMPDIR"' EXIT
    ++ mktemp -d
    + TMPDIR=/tmp/tmp.wdKSbrF3Pf
    + start
    + install_dependencies
    + type which
    + echo 检测运行环境
    检测运行环境
    + which apt-get
    + PM=apt
    + which yum
    + n=0
    + case $PM in
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^build-essential'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^clang'
    + '[' 2 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^flex'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^bison'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^g++'
    + '[' 4 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^gawk'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^gcc-multilib'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^g++-multilib'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^gettext'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^git'
    + '[' 3 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^libncurses-dev'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^libssl-dev'
    + '[' 1 -eq 0 ']'
    + for package_name in build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses-dev libssl-dev python3-distutils rsync unzip zlib1g-dev file wget make
    ++ apt list --installed
    ++ grep -c '^python3-distutils'
[... 873 lines cut ...]
    + exitstatus=0
    + '[' 0 -ne 6 ']'
    + break
    + echo end
    end
    + rm -rf /tmp/tmp.wdKSbrF3Pf
    ```

</details>

我现在也搞不清原因
试着在终端直接执行
```
TMPDIR=$(mktemp -d)
echo "$TMPDIR"
ls $TMPDIR
```

"mktemp -d"命令用于生成一个临时文件夹并返回文件夹的路径
实在不行可以尝试直接指定一个目录
把`TMPDIR=$(mktemp -d)`换成
```
TMPDIR="/tmp/config_build_tool_TMPDIR"
mkdir -p $TMPDIR
```

wanmyj, 2023-10-24:
No idea for the root cause but workaround works.

closed 2023-10-24 by wanmyj

wanmyj, 2023-10-26:
原来是Ubuntu自带的垃圾curl的问题，这么操作一下就OK了（操作完重启一个terminal）
```
$ sudo snap remove curl
$ sudo apt install curl
```