chenmozhijin/OpenWrt-K #48: 添加mosdns后提示缺少v2dat

yndzm, 2023-10-29:
Run TMPDIR=$(mktemp -d) || exit 1

WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'mosdns', which does not exist
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'v2dat', which does not exist
Error: 检查到软件包依赖问题,这极有可能导致编译错误，请前往build1检查软件包依赖步骤查看详情。
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'mosdns', which does not exist
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'v2dat', which does not exist

chenmozhijin, 2023-10-29:
你应该解决了吧 https://github.com/yndzm/OpenWrt-K/commit/30635686697f5f264fe777597a4dc1c54500871d

不过我看你[添加了docker](https://github.com/yndzm/OpenWrt-K/commit/fee2819078f9946af51341d8bf9c2c8a7ab9c223)，现在新版openwrt对docker的支持不是很好，因为openwrt已经把防火墙从iptables换成了nftables。

有一些方法可能可以提供帮助
https://github.com/openwrt/packages/issues/16818#issuecomment-1555652330

yndzm, 2023-10-29:
谢谢回复

closed 2023-11-04 by chenmozhijin