chenmozhijin/OpenWrt-K #49: 要怎样取消编译netdata？

smdx, 2023-10-30:
大佬你好，在config（luci.config、other.config）里面netdata相关的都前面加#或者=n，另外build-openwrt.yml里面netdata相关的都删除了，固件编译完成后还是带有netdata这个插件。请教一下要怎样才能取消编译这个插件？谢谢

chenmozhijin, 2023-10-30:
build-openwrt.yml里面netdata相关的并不需要删除，其实通过 https://github.com/smdx/OpenWrt-K/commit/6bf5234b4eaba8d3807fe2d65381be2eaf3e3b83 与 https://github.com/smdx/OpenWrt-K/commit/9c99196c19a56f2c850ec19a04b860a08b07e13f 这两个提交就已经取消编译netdata了，不过因为编译运行在 https://github.com/smdx/OpenWrt-K/commit/9c99196c19a56f2c850ec19a04b860a08b07e13f 之前，所以netdata仍然被编译了
[image]
[image]

smdx, 2023-10-30:
之前的fork记得netdata相关的config都#或者no还是编译进包，又乱七八糟的修改了些东西就丢弃修改重新Sync fork了。
今天这个编译任务是定时的,修改config后忘记看时间了...我再开个任务看看。谢谢了

closed 2023-10-31 by smdx