chenmozhijin/OpenWrt-K #55: 编译失败，重新克隆了你的库，也是失败

yndzm, 2023-11-15:
This step has been truncated due to its large size. Download the full logs from the  menu once the workflow run has completed.

chenmozhijin, 2023-11-15:
”“”
2023-11-15T07:32:41.5709120Z Finding targets
2023-11-15T07:32:41.5748583Z go: ../../go.mod requires go >= 1.21.4 (running go 1.21.2; GOTOOLCHAIN=local)
2023-11-15T07:32:41.5753994Z 
2023-11-15T07:32:41.5754295Z Building targets
2023-11-15T07:32:41.5797610Z go: ../../go.mod requires go >= 1.21.4 (running go 1.21.2; GOTOOLCHAIN=local)
2023-11-15T07:32:41.5804099Z 
2023-11-15T07:32:41.6849898Z make[2]: *** [Makefile:47: /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/xray-plugin-1.8.5/.built] Error 1
2023-11-15T07:32:41.6852332Z make[2]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/package/cmzj_package/openwrt-passwall-packages/xray-plugin'
2023-11-15T07:32:41.6856435Z time: package/cmzj_package/openwrt-passwall-packages/xray-plugin/compile#0.13#0.15#0.27
2023-11-15T07:32:41.6864132Z     ERROR: package/cmzj_package/openwrt-passwall-packages/xray-plugin failed to build.
2023-11-15T07:32:41.6867940Z make[1]: *** [package/Makefile:120: package/cmzj_package/openwrt-passwall-packages/xray-plugin/compile] Error 1
2023-11-15T07:32:41.6872163Z make[1]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt'
2023-11-15T07:32:41.6884855Z make: *** [/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/include/toplevel.mk:232: package/compile] Error 2
2023-11-15T07:32:41.6949868Z ##[error]The operation was canceled.
”“”

pallwall的依赖 xray-plugin要求go版本 >= 1.21.4，openwrt-23.05分支目前为1.21.3版本(23.05.0为1.21.2)

yndzm, 2023-11-16:
以前怎么能编译成功，现在不行了

chenmozhijin, 2023-11-16:
passwall 依赖 xray-plugin 更新了 https://github.com/xiaorouji/openwrt-passwall-packages/commit/dc45fc762d0d9728c030207b17bf6de36826c39d