chenmozhijin/OpenWrt-K #56: 请问怎样添加软件源仓库？

smdx, 2023-11-24:
想用ssr-plus和mosdns，编译时出现多个依赖提示which does not exist，在extpackages.config添加相关包又太多数量了。
要怎样添加软件源仓库？谢谢

依赖错误提示：
```
[19](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:20)
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'mosdns', which does not exist
[20](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:21)
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'v2dat', which does not exist
[21](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:22)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'lua-neturl', which does not exist
[22](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:23)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'mosdns', which does not exist
[23](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:24)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'v2dat', which does not exist
[24](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:25)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'shadow-tls', which does not exist
[25](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:26)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist
[26](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:27)
Error: 检查到软件包依赖问题,这极有可能导致编译错误，请前往build1检查软件包依赖步骤查看详情。
[27](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:28)
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'mosdns', which does not exist
[28](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:29)
WARNING: Makefile 'package/cmzj_package/luci-app-mosdns/Makefile' has a dependency on 'v2dat', which does not exist
[29](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:30)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'lua-neturl', which does not exist
[30](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:31)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'mosdns', which does not exist
[31](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:32)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'v2dat', which does not exist
[32](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:33)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'shadow-tls', which does not exist
[33](https://github.com/smdx/OpenWrt-K/actions/runs/6976303524/job/18984585510#step:23:34)
WARNING: Makefile 'package/cmzj_package/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist
```

chenmozhijin, 2023-11-25:
像添加软件包仓库一样添加就好了。

mosdns的两个依赖在仓库根目录，把拓展软件包路径设置为空即可(EXT_PACKAGES_PATH)

smdx, 2023-11-25:
谢谢，我修改试试。

closed 2023-11-26 by smdx