chenmozhijin/OpenWrt-K #58: 在extpackages里增加软件源却没有编译

missing233, 2024-01-28:
x86_64中的extpackages与defaultextpackages中都增加了
```
EXT_PACKAGES_NAME[16]="ipip6"
EXT_PACKAGES_PATH[16]=""
EXT_PACKAGES_REPOSITORIE[16]="https://github.com/missing233/openwrt-ipip6"
EXT_PACKAGES_BRANCH[16]=""
```
查看actions日志发现ipip6包只解压了，但是并没有被编译

chenmozhijin, 2024-01-29:
添加到extpackages只是代表OpenWrt-K会clone这个仓库并移动到对应的位置,你还需要在openwrt的配置文件中(那些后缀名为.config的文件)设置编译这个包

missing233, 2024-01-29:
> 添加到extpackages只是代表OpenWrt-K会clone这个仓库并移动到对应的位置,你还需要在openwrt的配置文件中(那些后缀名为.config的文件)设置编译这个包

有添加到配置文件中，但是仍然没有被编译

chenmozhijin, 2024-01-30:
把`EXT_PACKAGES_PATH[16]=""`设置成`EXT_PACKAGES_PATH[16]="net/ipv6/ipip6/"`试试

missing233, 2024-01-31:
> 把`EXT_PACKAGES_PATH[16]=""`设置成`EXT_PACKAGES_PATH[16]="net/ipv6/ipip6/"`试试 应该是文件夹层数太多了没有找到ipip6的Makefile

谢谢，已解决。

closed 2024-01-31 by missing233