chenmozhijin/OpenWrt-K #62: 大佬，我编译openwrt-23.05 分支，出现了一些问题

hexsen929, 2024-03-11:
https://github.com/openwrt/openwrt/issues/14841
大佬，我编译openwrt-23.05 分支，出现了一些问题，上面是我在openwrt提交的issues，但是他们居然让我问你。。
我们这边有办法解决这种问题吗？这个明显是它们更新了5.15.150内核后出现的问题。

chenmozhijin, 2024-03-11:
取消对`kmod-pf-ring`的编译应该就好了

修改[此行](https://github.com/hexsen929/OpenWrt-K/blob/2af595060e7452e513c214d292902ab6e8ee6521/config/x86_64/OpenWrt-K/compile.config#L2)

`kmod_compile_exclude_list=kmod-shortcut-fe-cm,kmod-shortcut-fe,kmod-fast-classifier,kmod-shortcut-fe-drv`
为
`kmod_compile_exclude_list=kmod-shortcut-fe-cm,kmod-shortcut-fe,kmod-fast-classifier,kmod-shortcut-fe-drv,kmod-pf-ring`

hexsen929, 2024-03-11:
> [此行](https://github.com/hexsen929/OpenWrt-K/blob/2af595060e7452e513c214d292902ab6e8ee6521/config/x86_64/OpenWrt-K/compile.config#L2)

感谢指点，我这边测试一下，我也准备用稳定发行版了，不然太折腾了

closed 2024-03-11 by hexsen929