chenmozhijin/OpenWrt-K #65: 大佬，好像files里的uci-default只有直接输入的命令生效了，if语句好像都没生效

m0eak, 2024-05-26:
[image]
[image]

chenmozhijin, 2024-05-26:
我重置试了一下好像没有什么问题
[image]

你可以试试

```sh
curl -sSL https://raw.githubusercontent.com/m0eak/OpenWrt-K/main/files/etc/uci-defaults/zzz-chenmozhijin -o /tmp/zzz-chenmozhijin && sh -x /tmp/zzz-chenmozhijin
```

来调试运行脚本(建议先进行备份)

m0eak, 2024-05-27:
> 我重置试了一下好像没有什么问题 ![屏幕截图 2024-05-26 234055](https://private-user-images.githubusercontent.com/110812055/333892260-9b88dbe9-3450-4e50-a527-d7504ee1323d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3NzYxODQsIm5iZiI6MTcxNjc3NTg4NCwicGF0aCI6Ii8xMTA4MTIwNTUvMzMzODkyMjYwLTliODhkYmU5LTM0NTAtNGU1MC1hNTI3LWQ3NTA0ZWUxMzIzZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyN1QwMjExMjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzJiMTE3Y2QyMjg0ZjllZTdjZTgxZDgzNDY1MGFjMWFjMWEzYWUwMTE2ZjM0NmYzMjE1NmRmNDcyMTc3MDhlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JAqqliWDKlc-6oXIDUYFap2BuS1wQJXUBogvdiTjVM4)
> 
> 你可以试试
> 
> ```shell
> curl -sSL https://raw.githubusercontent.com/m0eak/OpenWrt-K/main/files/etc/uci-defaults/zzz-chenmozhijin -o /tmp/zzz-chenmozhijin && sh -x /tmp/zzz-chenmozhijin
> ```
> 
> 来调试运行脚本(建议先进行备份)

直接手动运行是没问题的openclash配置也修改了，但就是刷入后，pve用img2kvm部署，所有的if都没运行，重置后也没用，下载你的release里的也是这样，很奇怪

chenmozhijin, 2024-05-27:
我尝试了最新的版本，无法复现此问题，你可以试试[OpenWrt-K 使用指南](https://github.com/chenmozhijin/OpenWrt-K/wiki)中的[安装固件](https://github.com/chenmozhijin/OpenWrt-K/wiki/%E5%9B%BA%E4%BB%B6%E4%BD%BF%E7%94%A8%E6%96%B9%E6%B3%95#2-%E5%AE%89%E8%A3%85%E5%9B%BA%E4%BB%B6)的方法(注意使用的是.vmdk结尾的镜像)
[image]
[image]
[image]

m0eak, 2024-05-28:
> 我尝试了最新的版本，无法复现此问题，你可以试试[OpenWrt-K 使用指南](https://github.com/chenmozhijin/OpenWrt-K/wiki)中的[安装固件](https://github.com/chenmozhijin/OpenWrt-K/wiki/%E5%9B%BA%E4%BB%B6%E4%BD%BF%E7%94%A8%E6%96%B9%E6%B3%95#2-%E5%AE%89%E8%A3%85%E5%9B%BA%E4%BB%B6)的方法(注意使用的是.vmdk结尾的镜像) ![屏幕截图 2024-05-27 224656](https://private-user-images.githubusercontent.com/110812055/334142497-ec5d15b4-773e-4775-b2e0-e2b4d93be890.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4ODcwMTYsIm5iZiI6MTcxNjg4NjcxNiwicGF0aCI6Ii8xMTA4MTIwNTUvMzM0MTQyNDk3LWVjNWQxNWI0LTc3M2UtNDc3NS1iMmUwLWUyYjRkOTNiZTg5MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyOFQwODU4MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDBkMzcxMTJlZmVmZDBhZGU0MjdjMGNiYTYyZDdlNmM4OTdhMWQ3ZmU2MTVhOGU1YTBkNzNhNjM3OGVlNTk1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5wMGdA5MEmNXxmdFTbEXq_Pi6ZqRCBvYvRuGkm2fDFg) ![屏幕截图 2024-05-27 224815](https://private-user-images.githubusercontent.com/110812055/334142512-f3a1a1e4-f58b-4bfe-9832-875a441e7f16.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4ODcwMTYsIm5iZiI6MTcxNjg4NjcxNiwicGF0aCI6Ii8xMTA4MTIwNTUvMzM0MTQyNTEyLWYzYTFhMWU0LWY1OGItNGJmZS05ODMyLTg3NWE0NDFlN2YxNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyOFQwODU4MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YmFkYzdkNGYxNzhhMGY4NWE4YTMwMzZiNjkzZDk3ZjA3Y2E4N2NkODk0ZjBlOWQ2MmMwNDU5OTZkZmU4MDE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iZREWkHG6Fl5T2QN39iox0yd3flXrkJG6CuMNzD80ts) ![屏幕截图 2024-05-27 224710](https://private-user-images.githubusercontent.com/110812055/334142522-423c1f22-6af3-4727-b3f8-777324853166.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4ODcwMTYsIm5iZiI6MTcxNjg4NjcxNiwicGF0aCI6Ii8xMTA4MTIwNTUvMzM0MTQyNTIyLTQyM2MxZjIyLTZhZjMtNDcyNy1iM2Y4LTc3NzMyNDg1MzE2Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyOFQwODU4MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MzcwMzUxNzJlNWJhNGI0NzVkNGQyYjg3Zjk2NWZkZjAzODNhNDY2MTBhOGZiOWIyNTc0NTU3NTk4ZjY4NWEzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D2yJYFGeRCOWQo0URaX4NBBwk4n2QQ3axSOfYYP1Cq4)

用这个方法确实有了

closed 2024-05-28 by chenmozhijin

m0eak, 2024-05-31:
我把我fork的另外建立了一个init.sh然后uci default sh +x运行，也不行，然后放最后sleep 5秒再执行就好了