chenmozhijin/OpenWrt-K #80: 想添加第三方插件和修改OPKG的配置

LIUBANGJIAN, 2025-01-23:
大佬，你好。
我想添加luci-app-alist和luci-app-lucky，要怎么添加？我修改了[default-extpackages.config](https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/default-extpackages.config)和[luci.config](https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/luci.config)后，重新编译后，固件没有这两个插件，请问是设置不对吗？还是设置的位置不对？

还有我想修改OPKG的配置，改成国内的库，请问在编译前要怎么修改呢？

北大的库https://mirrors.pku.edu.cn/openwrt/releases/23.05.5/targets/x86/64/packages

chenmozhijin, 2025-01-23:
修改https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/OpenWrt-K/extpackages.config而不是修改[default-extpackages.config](https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/default-extpackages.config)

将`sed -i 's_https\?://downloads.openwrt.org_https://mirrors.tuna.tsinghua.edu.cn/openwrt_' /etc/opkg/distfeeds.conf`添加到https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/files/etc/uci-defaults/zzz-chenmozhijin 的`exit 0`之前

LIUBANGJIAN, 2025-01-24:
大佬，没有效果。
https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/OpenWrt-K/extpackages.config
我依葫芦画瓢加了下面内容
EXT_PACKAGES_NAME[28]="luci-app-alist"
EXT_PACKAGES_PATH[28]="luci-app-alist"
EXT_PACKAGES_REPOSITORIE[28]="https://github.com/sbwml/luci-app-alist"
EXT_PACKAGES_BRANCH[28]=""
EXT_PACKAGES_NAME[29]="luci-app-lucky"
EXT_PACKAGES_PATH[29]="luci-app-lucky"
EXT_PACKAGES_REPOSITORIE[29]="https://github.com/gdy666/luci-app-lucky"
EXT_PACKAGES_BRANCH[29]=""

https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/luci.config
CONFIG_PACKAGE_luci-app-zerotier=n
CONFIG_PACKAGE_luci-app-wireguard=y
CONFIG_PACKAGE_luci-app-alist=y
CONFIG_PACKAGE_luci-app-lucky=y
# end of 3. Applications

https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/files/etc/uci-defaults/zzz-chenmozhijin
/etc/init.d/sing-box disable
sed -i 's_https\?://downloads.openwrt.org_https://mirrors.tuna.tsinghua.edu.cn/openwrt_' /etc/opkg/distfeeds.conf
exit 0

编译出来后没有添加相关的插件和OPKG的配置也没有修改，

还有编译的有提示错误
Annotations
2 errors and 1 warning
构建软件包(x86_64)
发生错误: CalledProcessError: Command '['make', 'package/install', '-j1', 'V=s']' returned non-zero exit status 2.
构建软件包(x86_64)
Process completed with exit code 1.
构建软件包(x86_64)
No files were found with the provided path: /home/runner/work/OpenWrt-K/OpenWrt-K/tmp/tmp60pxybar/openwrt.tar.gz. No artifacts will be uploaded.

还有个问题，升级固件提示错误，无法升级。
0,No error information 关闭
上传请求失败：I/O failure while writing temporary file 关闭

chenmozhijin, 2025-01-24:
`EXT_PACKAGES_PATH[29]="luci-app-lucky"`改成`EXT_PACKAGES_PATH[29]=""`因为[luci-app-lucky](https://github.com/gdy666/luci-app-lucky)仓库里不仅有[luci-app-lucky](https://github.com/gdy666/luci-app-lucky/tree/main/luci-app-lucky)还有它的依赖[lucky](https://github.com/gdy666/luci-app-lucky/tree/main/lucky),设置`EXT_PACKAGES_PATH[29]="luci-app-lucky"`就只会添加luci-app-lucky而不添加它的依赖lucky导致编译失败

不是没有修改，是编译失败了，你下载的是原来的旧版本

升级固件提示错误可能是固件太大了，内存不太够，如果是虚拟机可以直接替换硬盘

LIUBANGJIAN, 2025-01-24:
大佬，还是编译失败，麻烦看看。

Annotations
2 errors and 1 warning
构建软件包(x86_64)
发生错误: CalledProcessError: Command '['make', 'package/compile', '-j1', 'V=s']' returned non-zero exit status 2.
构建软件包(x86_64)
Process completed with exit code 1.
构建软件包(x86_64)
No files were found with the provided path: /home/runner/work/OpenWrt-K/OpenWrt-K/tmp/tmpvlpmwn9_/openwrt.tar.gz. No artifacts will be uploaded.

make[2]: *** [Makefile:99: /home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt/build_dir/target-x86_64_musl/alist-3.41.0/.built] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt/package/cmzj_packages/luci-app-alist/alist'
time: package/cmzj_packages/luci-app-alist/alist/compile#5.02#2.15#4.51
    ERROR: package/cmzj_packages/luci-app-alist/alist failed to build.
make[1]: *** [package/Makefile:129: package/cmzj_packages/luci-app-alist/alist/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt'
make: *** [/home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt/include/toplevel.mk:233: package/compile] Error 2
Error: 2025-01-24 16:39:14,316- openwrt(79) - make:编译失败，请检查错误信息
Error: 2025-01-24 16:39:14,316- __main__(65) - <module>:发生错误
Traceback (most recent call last):
  File "/opt/OpenWrt-K/build_helper/__main__.py", line 63, in <module>
    main()
  File "/opt/OpenWrt-K/build_helper/__main__.py", line 49, in main
    build_packages(config)
  File "/opt/OpenWrt-K/build_helper/build.py", line 160, in build_packages
    openwrt.make("package/compile")
  File "/opt/OpenWrt-K/build_helper/utils/openwrt.py", line 77, in make
    self.make(target, debug=True)
  File "/opt/OpenWrt-K/build_helper/utils/openwrt.py", line 80, in make
    raise subprocess.CalledProcessError(result.returncode, result.args, result.stdout, result.stderr)
subprocess.CalledProcessError: Command '['make', 'package/compile', '-j1', 'V=s']' returned non-zero exit status 2.
[INFO]2025-01-24 16:39:14,319- __main__(66) - <module>:开始收集错误信息...
[INFO]2025-01-24 16:39:14,320- __main__(84) - <module>:正在打包 openwrt 文件夹...
[DEBUG]2025-01-24 17:05:51,904- upload(30) - add:Add Artifact: build-packages-x86_64-openwrt-1737709551.9049406 /home/runner/work/OpenWrt-K/OpenWrt-K/tmp/tmpvlpmwn9_/openwrt.tar.gz
[DEBUG]2025-01-24 17:05:53,807- upload(30) - add:Add Artifact: build-packages-x86_64-errorinfo-1737709553.8077013 /home/runner/work/OpenWrt-K/OpenWrt-K/errorinfo
[DEBUG]2025-01-24 17:05:53,807- upload(55) - save:Save UpLoad Action file to /opt/OpenWrt-K/.github/action/upload/action.yml
Error: 发生错误: CalledProcessError: Command '['make', 'package/compile', '-j1', 'V=s']' returned non-zero exit status 2.
Error: Process completed with exit code 1.

chenmozhijin, 2025-01-24:
编译alist需要运行器上有libfuse-dev加上应该就好了 e4da5501ffc4a5503e363e3e4656fc8fadba1ee9

closed 2025-01-25 by chenmozhijin

LIUBANGJIAN, 2025-01-26:
> `EXT_PACKAGES_PATH[29]="luci-app-lucky"`改成`EXT_PACKAGES_PATH[29]=""`因为[luci-app-lucky](https://github.com/gdy666/luci-app-lucky)仓库里不仅有[luci-app-lucky](https://github.com/gdy666/luci-app-lucky/tree/mai [...]

大佬，我按照上面的方法现在编译正常了。也opkg的配置也修改了。现在就是不能直接img升级固件，老是提示错误，各个大小我都试过了，都是更新报错。用的是爱快虚拟机，8核8g内存，如果每次更新都替换硬盘的话，每次都要重新配置，这样不太方便。

https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/image.config

CONFIG_TARGET_KERNEL_PARTSIZE=32 试过64、512
CONFIG_TARGET_ROOTFS_PARTSIZE=1024 试过512

都是不正常img文件更加，请问是什么原因？请问怎么解决呢？

chenmozhijin, 2025-01-26:
你可以替换前先备份，替换后在恢复(在[`系统->备份与升级`](http://192.168.1.1/cgi-bin/luci/admin/system/flash)中，不过跟保留配置升级一样，可能容易出现bug)。

img升级固件会把整个固件拷贝到内存中(\tmp目录),对虚拟机的内存要求比较大