chenmozhijin/OpenWrt-K #81: 大佬准备什么时候支持24版本呀

hexsen929, 2025-02-06:
😬

chenmozhijin, 2025-02-06:
https://github.com/chenmozhijin/OpenWrt-K/tree/v24.10.0

在做了

chenmozhijin, 2025-02-06:
ba7c955b430524c9921f697702ca4bfa4b41c000 merged

不过我删了一些包passwall、passwall2、luci-app-rclone、luci-app-ddns、luci-app-aria2，需要的话可以修改编译配置加回来

closed 2025-02-06 by chenmozhijin

5points, 2025-02-15:
想问一下，为什么24.10的空间不像之前一样为4G？
实际上ddns和passwall（1）挺有用的，可以加回来吗？
*因为大佬你编译的我直接就可以用，挺方便的。这里并没有什么个人订制的需求（就不用修改config再编译一次了）。

chenmozhijin, 2025-02-20:
> 想问一下，为什么24.10的空间不像之前一样为4G？
> 实际上ddns和passwall（1）挺有用的，可以加回来吗？
> *因为大佬你编译的我直接就可以用，挺方便的。这里并没有什么个人订制的需求（就不用修改config再编译一次了）。

因为passwall太大了，加上它固件体积就翻了一倍，我现在主用openclash用不上它所以就把它去掉了。

passwall有提供安装包：
[passwall](https://github.com/xiaorouji/openwrt-passwall/releases/latest)
[passwall2](https://github.com/xiaorouji/openwrt-passwall2/releases/latest)

你可以在 `系统->软件包` 中上传软件包来安装。

ddns可以在 `系统->软件包` 中先更新列表,再在过滤器里面输入`ddns`
安装:
luci-app-ddns
luci-i18n-ddns-zh-cn
和与你dns提供商对应的脚本