chenmozhijin/OpenWrt-K #83: 使用PVE的运行镜像，在重启时会概率遇到丢失配置的问题

5points, 2025-02-08:
你好
我在两个迷你主机用不同的方式运行系统时，会出现一些问题
1.在PVE以Qemu运行此固件，在重启Qemu虚拟机的时候，Openwrt会概率丢失配置。这时候得重新导入镜像才可以运作。
2.在PVE以LXC 运行此固件，在多天没重启系统时，发现LXC的储存容量爆满。导致无法访问页面。导致不得不重装。
第一个例子用的是N100的CPU，这个情况我遇到很多次。
第二个例子用的是J1900的环境

chenmozhijin, 2025-02-09:
试试[固件使用方法](https://github.com/chenmozhijin/OpenWrt-K/wiki/%E5%9B%BA%E4%BB%B6%E4%BD%BF%E7%94%A8%E6%96%B9%E6%B3%95)中的安装方法

5points, 2025-02-09:
谢谢你，我这次改用bios模式（非UEFI）来启动了，先用一段时间试试看。我猜估计是我这里UEFI的设置有问题（我有在pve添加EFI磁盘），会不会跟我“添加EFI磁盘”这个设定有关，导致我概率重启虚拟机就会丢配置（恢复出厂设置）？
额外再请教一下，为什么教程是改用local来运行qcow镜像，且不用“添加EFI磁盘”。
在1900那个环境，使用虚拟机会很卡。不得已才使用LXC容器运行。
出现（LXC）储存容量饱满的情况是第一次。请问你有遇到过吗？

chenmozhijin, 2025-02-09:
储存容量爆满可能是日志文件导致的，我有一次是AdGuard Home的查询日志保留时间开的90天导致爆满，但还能连上

这是我目前的分配的硬件

[image]
使用的虚拟硬盘为openwrt-x86-64-generic-squashfs-combined-efi.vmdk，不用“添加EFI磁盘”一样可用正常启动

重启虚拟机就会丢配置可能跟[OverlayFS](https://openwrt.org/docs/techref/filesystems#overlayfs)文件系统有关，[OverlayFS](https://openwrt.org/docs/techref/filesystems#overlayfs)由一个只读文件系统[SquashFS](https://openwrt.org/docs/techref/filesystems#squashfs)，与一个可写文件系统[JFFS2](https://openwrt.org/docs/techref/filesystems#jffs2)（挂载到/overlay）组成，当/overlay分区爆满时就会出现丢配置的问题，你可用在系统-挂载点中或通过`df -h`目录查看各个挂载点的大小

[image]

如果怕丢配置可以使用ext4文件系统的虚拟硬盘，不过会无法使用恢复到出厂设置

closed 2025-02-11 by 5points