chenmozhijin/OpenWrt-K #84: 构建软件包报错

LIUBANGJIAN, 2025-02-20:
近期编译都成功，昨晚开始突然不行了。没做调整，是上游有什么变动吗？大佬麻烦帮忙看看。
https://github.com/LIUBANGJIAN/OpenWrt-K/actions/runs/13421660811/job/37496645246

chenmozhijin, 2025-02-20:
lucky的问题
```
* pkg_run_script: package "lucky" preinst script returned status 1.
 * preinst_configure: Aborting installation of lucky.
 * opkg_install_cmd: Cannot install package luci-app-lucky.
 * pkg_run_script: package "lucky" preinst script returned status 1.
 * preinst_configure: Aborting installation of lucky.
 * opkg_install_cmd: Cannot install package lucky.
```

LIUBANGJIAN, 2025-02-20:
> lucky的问题
>
> ```
> * pkg_run_script: package "lucky" preinst script returned status 1.
>  * preinst_configure: Aborting installation of lucky.
>  * opkg_install_cmd: Cannot install package luci-app-lucky.
>  * [...]

有添加的，放到最后，几天前还可以编译。
https://github.com/LIUBANGJIAN/OpenWrt-K/blob/main/config/x86_64/OpenWrt-K/extpackages.config

closed 2025-02-22 by chenmozhijin