chenmozhijin/OpenWrt-K #85: 大佬空了帮忙看看呢，xray-core昨天更新后编译失败了

hexsen929, 2025-03-04:
make[2] -C feeds/packages/lang/perl-parse-yapp host-compile
    ERROR: package/cmzj_packages/hexsen929/xray-core failed to build.
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt/include/toplevel.mk:233: package/compile] Error 1
Error: 2025-03-05 01:22:06,248- openwrt(76) - make:编译失败，尝试使用debug模式重新编译
[DEBUG]2025-03-05 01:22:06,249- openwrt(72) - make:运行命令：make package/compile -j1 V=s
make[2]: Entering directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workdir/openwrt/scripts/config'

hexsen929, 2025-03-05:
好像是xray-core使用go1.24的原因，我这边再研究下。

chenmozhijin, 2025-03-05:
应该是的

hexsen929, 2025-03-05:
> 应该是的

嗯嗯，过了，改成24就可以了，感谢🙏

closed 2025-03-05 by hexsen929

LIUBANGJIAN, 2025-03-13:
> > 应该是的
> 
> 嗯嗯，过了，改成24就可以了，感谢🙏

请问在什么地方修改呢？

chenmozhijin, 2025-03-13:
> > > 应该是的
> > 
> > 
> > 嗯嗯，过了，改成24就可以了，感谢🙏
> 
> 请问在什么地方修改呢？
95dd920e04a3855167f7d5e0c432cbcf57ec2709